ea4k/klog #1: ADIF import should mark online services as to be sent

ea4k, 2021-01-07:
LoTW, qrz.com, Clublog, eQSL.cc, ...

barjac, 2021-01-10:
If an adif has just been imported from <your favourite online service> why would you want to automatically send it back?
If the records just imported were 'flagged' in some easily visible way, and options were available to 'mark flagged records to be uploaded to ...' etc. etc. it would be much more versatile. I think that a general purpose flag/tag whatever it may be called would be useful for these tasks.

ea5wa, 2021-01-14:
If the ADIF file being imported has QSOs marked as "sended" are leave it like this.

Imported QSOs without any information about "sended to online service" are marked correctly to be sended using KLog

closed 2021-01-31 by ea4k