ea4k/klog #10: Download LoTW just the new qsl not the full log.

ea4k, 2021-01-10:


ea5wa, 2025-08-08:
Actually, it takes so long (half an hour for a 13.000 QSOs file in my case). Even if you had downloaded the day before.....

In my case, I have 8.667 QSLs in LoTW, but when I use KLog to download, the downloaded file has 13.000 QSOs

[image]

[image]

I don't know where this 13.000 QSOs come from.....

Is not possible to download just the confirmations?

ea5wa, 2025-08-08:
Downloading lotwreport.adi from LoTW web page, the file has 7.673 QSOs

[image]