ea4k/klog #11: Add a check box on main display to disable status info from wsjtx

barjac, 2021-01-10:
When wsjtx is running it impossible to edit a record or manually enter a record without it being overwritten or deleted by data coming from wsjtx.
It is inconvenient to stop wsjtx and impossible to use klog settings to turn off status info, as doing so clears the edit in progress.

ea5wa, 2021-02-15:
Doesn't it solved unchecking the third checkbox in Setup-WSJTX? Update status from WSJT

barjac, 2021-05-12:
On 15/02/2021 17:50, ea5wa wrote:
> Doesn't it solved unchecking the third checkbox in Setup-WSJTX? Update 
> status from WSJT
> 
> 

No, because this is needed instantly.

Entering 'Setup' for any reason wipes any data already entered.

barjac, 2021-05-18:
To clarify this issue further explanation of the use case that causes problems is needed:
A station is using a wsjtx mode on HF and has qrz.com lookup and hamlib running.
The current callsign in wsjtx is controlling the 'callsign' field in klog.
Hamlib (read only) is setting the Mode, Band and frequency.
QRZ.com is entering the Name, City and Locator.
Now a DX CQ call is heard on a different radio, maybe VHF, and is worked.
In order to log this call, wsjtx, QRZ.com auto-lookups and hamlib would all have to be disabled in setup.
As you can imagine this is impractical.

ea4k, 2022-01-19:
Solved in master

closed 2022-01-19 by ea4k