ea4k/klog #14: Imported QSOs using ADIF without "my_gridsquare" can not be uploaded to LoTW

ea5wa, 2021-01-12:
After Importing QSOs with "my_gridsquare" field empty, these can not be uploaded to loTW because they don't appear in the QSOs to be uploaded window. Once this field (MY_GRIDSQUARE) is fullfilled, they appear

ea5wa, 2021-01-13:
Not possible  to reproduced it..... Is working fine

ea5wa, 2021-01-17:
The problem has appeared again..... Only happens with imported QSO (from an ADIF file without "MY_GRIDSQUARE" data)

ea4k, 2022-01-13:
When a call is enteres, the mainwindow assumed that there is always something coming from qrz.com and clears the data that is expecting to come.

We should replace without cleaning before

ea5wa, 2022-01-15:
Importing an adif without "my_gridsquare" field, QSOs are saved in the database without any info (empty field) in "lotw_qsl_sent". Then, if you try to Upload these QSOs to LoTW, the QSOs don't appear in the Export Widget, because lotw_qsl_sent is not "Q" (is empty).
eQSL Tab data shows "Q" but in reality is empty

[image]

ea5wa, 2022-01-15:
When editing the QSO, even without modify any field, after save changes, "Q" is written in the database and now appears in the Export Widget

ea4k, 2022-02-11:
This "error" is cause when importing ADIF without the proper information on LoTW to take an action... We may either add a "Q" by default to all the QSOs without that information or ask the user what to do.
Any suggestion?

I will remove the "Bug" tag and add the "Enhancement" tag.

closed 2025-08-13 by ea4k