ea4k/klog #36: QTextStreamFunctions::endl seems to be deprecated.

ea4k, 2021-01-31:
setupdialog.cpp:706:50: warning: 'QTextStream& QTextStreamFunctions::endl(QTextStream&)' is deprecated: Use Qt::endl [-Wdeprecated-declarations]
  706 |                  stream << "QRZcomAuto=True;" << endl;
      |

barjac, 2021-02-01:
`In file included from /usr/include/qt5/QtCore/qdebug.h:49,
                 from /usr/include/qt5/QtCore/qcborcommon.h:45,
                 from /usr/include/qt5/QtCore/qcborvalue.h:45,
                 from /usr/include/qt5/QtCore/qcborarray.h:43,
                 from /usr/include/qt5/QtCore/QtCore:38,
                 from /usr/include/qt5/QtWidgets/QtWidgetsDepends:3,
                 from /usr/include/qt5/QtWidgets/QtWidgets:3,
                 from main.cpp:18:
/usr/include/qt5/QtCore/qtextstream.h:293:75: note: declared here
  293 | Q_CORE_EXPORT QT_DEPRECATED_VERSION_X(5, 15, "Use Qt::endl") QTextStream &endl(QTextStream &s);
      |                                                                           ^~~~
main.cpp:67:62: warning: 'QTextStream& QTextStreamFunctions::endl(QTextStream&)' is deprecated: Use Qt::endl [-Wdeprecated-declarations]
   67 |             cout << "     -?           Display this help" << endl;
      |                                                              ^~~~
In file included from /usr/include/qt5/QtCore/qdebug.h:49,
                 from /usr/include/qt5/QtCore/qcborcommon.h:45,
                 from /usr/include/qt5/QtCore/qcborvalue.h:45,
                 from /usr/include/qt5/QtCore/qcborarray.h:43,
                 from /usr/include/qt5/QtCore/QtCore:38,
                 from /usr/include/qt5/QtWidgets/QtWidgetsDepends:3,
                 from /usr/include/qt5/QtWidgets/QtWidgets:3,
                 from main.cpp:18:
/usr/include/qt5/QtCore/qtextstream.h:293:75: note: declared here
  293 | Q_CORE_EXPORT QT_DEPRECATED_VERSION_X(5, 15, "Use Qt::endl") QTextStream &endl(QTextStream &s);
      |                                                                           ^~~~
main.cpp:67:62: warning: 'QTextStream& QTextStreamFunctions::endl(QTextStream&)' is deprecated: Use Qt::endl [-Wdeprecated-declarations]
   67 |             cout << "     -?           Display this help" << endl;
      |                                                              ^~~~
In file included from /usr/include/qt5/QtCore/qdebug.h:49,
                 from /usr/include/qt5/QtCore/qcborcommon.h:45,
                 from /usr/include/qt5/QtCore/qcborvalue.h:45,
                 from /usr/include/qt5/QtCore/qcborarray.h:43,
                 from /usr/include/qt5/QtCore/QtCore:38,
                 from /usr/include/qt5/QtWidgets/QtWidgetsDepends:3,
                 from /usr/include/qt5/QtWidgets/QtWidgets:3,
                 from main.cpp:18:
/usr/include/qt5/QtCore/qtextstream.h:293:75: note: declared here
  293 | Q_CORE_EXPORT QT_DEPRECATED_VERSION_X(5, 15, "Use Qt::endl") QTextStream &endl(QTextStream &s);
      |                                                                           ^~~~
main.cpp:68:62: warning: 'QTextStream& QTextStreamFunctions::endl(QTextStream&)' is deprecated: Use Qt::endl [-Wdeprecated-declarations]
   68 |             cout << "     -h           Display this help" << endl;
      |                                                              ^~~~
In file included from /usr/include/qt5/QtCore/qdebug.h:49,
                 from /usr/include/qt5/QtCore/qcborcommon.h:45,
                 from /usr/include/qt5/QtCore/qcborvalue.h:45,
                 from /usr/include/qt5/QtCore/qcborarray.h:43,
                 from /usr/include/qt5/QtCore/QtCore:38,
                 from /usr/include/qt5/QtWidgets/QtWidgetsDepends:3,
                 from /usr/include/qt5/QtWidgets/QtWidgets:3,
                 from main.cpp:18:
/usr/include/qt5/QtCore/qtextstream.h:293:75: note: declared here
  293 | Q_CORE_EXPORT QT_DEPRECATED_VERSION_X(5, 15, "Use Qt::endl") QTextStream &endl(QTextStream &s);
      |                                                                           ^~~~
main.cpp:68:62: warning: 'QTextStream& QTextStreamFunctions::endl(QTextStream&)' is deprecated: Use Qt::endl [-Wdeprecated-declarations]
   68 |             cout << "     -h           Display this help" << endl;
      |                                                              ^~~~
In file included from /usr/include/qt5/QtCore/qdebug.h:49,
                 from /usr/include/qt5/QtCore/qcborcommon.h:45,
                 from /usr/include/qt5/QtCore/qcborvalue.h:45,
                 from /usr/include/qt5/QtCore/qcborarray.h:43,
                 from /usr/include/qt5/QtCore/QtCore:38,
                 from /usr/include/qt5/QtWidgets/QtWidgetsDepends:3,
                 from /usr/include/qt5/QtWidgets/QtWidgets:3,
                 from main.cpp:18:
/usr/include/qt5/QtCore/qtextstream.h:293:75: note: declared here
  293 | Q_CORE_EXPORT QT_DEPRECATED_VERSION_X(5, 15, "Use Qt::endl") QTextStream &endl(QTextStream &s);
      |                                                                           ^~~~
main.cpp:69:68: warning: 'QTextStream& QTextStreamFunctions::endl(QTextStream&)' is deprecated: Use Qt::endl [-Wdeprecated-declarations]
   69 |             cout << "     -v           Display program version" << endl;
      |                                                                    ^~~~`

barjac, 2021-02-01:
I tried to add example guild output but the code markup here is useless so link instead.
[http://mtf.duckdns.org/pub/linux/barjac/distrib/cauldron/x86_64/log/klog-1.4.6-1.mga8_bcj.src.rpm/build.x86_64.0.20210130233336.log](url)

ea4k, 2021-08-06:
Linked to #323

closed 2021-08-06 by ea4k