ea4k/klog #110: File sats.dat is not properly updated

ea4k, 2021-04-01:
Tried creating an empty file and selecting it for update but on selection of the file the dialog just closes.

ea5wa, 2022-01-24:
[image]
I'm just testing it, are you refering to "Import Button"?

ea5wa, 2022-01-24:
In my case, dialog is not close, and remains as you can see in the image, until click in "OK"

ikbenkous, 2023-11-25:
I get ea4k's behavior: I select an empty file, the dialog closes and the file remains empty. No idea what it's supposed to do.

ea5wa, 2023-12-15:
I can't reproduce that but. I download sats.dat (from my github) and satellites list is updated correctly. Wich sats.dat are you using?

ea4k, 2023-12-17:
I am actively working on #121, to download.
I will afterwards review the parsing and update of info in KLog.

closed 2025-04-17 by ea4k