ea4k/klog #117: Installation of /usr/tests does not respect BUILDROOT

hobbes1069, 2021-04-09:
I am working on updating the Fedora package but for packaging purposes everything is installed into a BUILDROOT and installation is done as a normal user which works for the main components:
```
/usr/lib64/qt5/bin/qmake -install qinstall -exe build/target/klog /builddir/build/BUILDROOT/klog-1.5-1.fc32.x86_64/usr/bin/klog
/usr/lib64/qt5/bin/qmake -install qinstall /builddir/build/BUILD/klog-1.5/src/Changelog /builddir/build/BUILDROOT/klog-1.5-1.fc32.x86_64/usr/share/klog/Changelog
/usr/lib64/qt5/bin/qmake -install qinstall /builddir/build/BUILD/klog-1.5/src/COPYING /builddir/build/BUILDROOT/klog-1.5-1.fc32.x86_64/usr/share/klog/COPYING
```
But not for tests:
```
make[2]: Entering directory '/builddir/build/BUILD/klog-1.5/tests/utilities'
mkdir: cannot create directory '/usr/tests': Permission denied
```

ea4k, 2021-05-01:
Tests are not to be packaged for users as they are developers tests only.
Only the content in the "src" folder is to be packaged for users.

ikbenkous, 2023-11-25:
I just saw a bunch test stuff installed in my system's /usr/tests/. Is this a bug or expected behavior?

ea4k, 2026-03-14:
Fixed.