ea4k/klog #265: [Enhancement] - Tab key behaviour in the main window

foldynl, 2021-06-09:
It would be nice to improve the behavior of the tab key in the main window. Currently, if a callsign is inserted and then Tab key is press then a focus (cursor) is switched to band and then Mode then Time. 

The nature sequence of cursor focusing after pressing tab should be:
Callsign->Time (only in case when real-time is not checked otherwise skip it)->RST(tx)->RST(rx)->QTH->Locator->Name-> the rest of fields.

It is not necessary to focus a split checkbox by pressing Tab key.

ea5wa, 2021-06-10:
I think will be nice to have the possibility to modify this sequence by the user

foldynl, 2021-06-11:
It is also a possibility. But the question is whether you want an user to have such freedom. In that case, it would be nice that the user will have a freedom in what he/she sees in QSO log (column selection and order), what font is used, etc. I'm not a fan that the user can set everything - most of users leave the setting in default (when the design is good) and this freedom is redeemed by the fact that the setup dialog will contain many options 

I would be inspired by for example N1MM where the jump sequence is given by the contest. Therefore, the sequence should reflect a common QSO sequence.

ea4k, 2021-06-26:
It is better to start small,  not doing it user configurable.
Following N1MM makes sense for me.

ea4k, 2021-07-13:
What do you think about this:
QSO->band->Mode->Date/time (if not auto)->RSTtx->RSTrx->QTH->Name->Grid->Comment

No sequence defined after that as there are too many possibilities.

foldynl, 2021-07-13:
I think that it is not important to jump to band and mode for every QSO because an operator does not switch band or mode so often. I would prefer to totally skip band and mode and focus only on the things important for QSO - RST/QTH/Name/Grid/Comment. Another interesting thing is that the input of data are different, whether it is HF or VHF / UHF QSO. For example, Grid is not commonly exchanged information in HF QSOs - except digital modes where grid is exchanged. On the other hand a grid is an essential information during VHF/UHF QSO. 
Therefore, it is a good starting point to have your order with modification like this: QSO->Data/time (if not auto)->RSTtx->RSTrx->QTH->Name->Grid->Comment. Maybe an order QSO->Data/time (if not auto)->RSTtx->RSTrx->Name->QTH->Grid->Comment would be more natural.