ea4k/klog #267: HamLib setting does not support radios which are connected via network

foldynl, 2021-06-09:
Unfortunately, there is no way how to define HamLib radio which is connected via network (in Setup->HamLib)

For example:

In case of [NET rigctl], it is necessary to define IP address. Unfortunately, the Port field is a non-editable dropdown where  only local ports (USBs, COMs) are present.

ea4k, 2021-06-12:
Do you have any usecase to test NET rigctl?
I will add support but I don't know how to test it.

foldynl, 2021-06-13:
That's good question how to test it. 

I use the rigctl daemon with my IC7300. The command is $RIGCTLD -m 3073 -r  /dev/serial/by-id/usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controller_IC-7300_03018006-if00-port0 -s 115200 --set-conf=data_bits=8,stop_bits=2,serial_parity=None,serial_handshake=None  -v. But it expects that you have a rig connected to your computer. 

On the other hands, you can use just simple rigctld -v (without -m) what means that a dummy rig will be used. It is good starting point for your tests.

Just one note. Not all rigs in hamlib can be connected via network. You can use rig's capabilities (struct rig_caps attribute port_type) where port type can be RIG_PORT_SERIAL or RIG_PORT_NETWORK. Based on the type, you can determine whether the rig need an IP address or serial port address.

closed 2021-06-17 by ea4k

foldynl, 2021-06-17:
Thank you...looks much better. Unfortunately, I have found an issue related to config file. I will open a new issue for it.

commit faaad968 (2021-06-14): Hamlib network radios (closes #267 )