ea4k/klog #268: [Enhancement] - Improve QSO time

foldynl, 2021-06-09:
The main window -> Date&Time fields are followed by check button, unfortunately, there is no description what exactly the check button does.

[image]

The label may not be used when, for example, following buttons will be used:
[image]
[image]

ea4k, 2021-06-09:
I have added a label.
I am not really sure if it is redundant or not.

[image]

What do you think?

ea4k, 2021-06-09:
Added

closed 2021-06-09 by ea4k

foldynl, 2021-06-10:
Thanks..much better. 
To be hones, I would prefer to use an icons - it does not burden translators who have to constantly update their translations. Otherwise, I am thinking about the meaning of "real time". I know what you want to say but "real time" has another meaning in IT terminology  (https://en.wikipedia.org/wiki/Real-time). For example, in case of "Send each QSO to ClubLog in real time" it would be better to use "Send each QSO to ClubLog immediately". Unfortunately, I am not sure what should be used in case of "real time" beside date&time. Therefore, the button + icon is better solution. I will try to look at the source as it is implemented there.