ea4k/klog #295: [Enhancement] - Reworking Configuration File

foldynl, 2021-06-18:
what do you think about my crazy idea to rework how the configuration file is load and save and how configuration parameters are used. My idea is to remove currently used configuration procedures and replace them by a common QSettings class.

As I can see, there are two places where the Klog configuration file is processed - mainwindow and setupdialog. Several functions are very similar and it is possible to make a simple mistake here. I would like to replace this by storing the configuration parameters using the QSettings class. If any parameter will be needed, it will be extracted from the configuration file by QSettings class.

I can see few problems
- the change will be quite big. Many files will be impacted. Not easy to merge it.
- at this moment, I cannot image how to migrate from old config format to new config format

What do you thing about the idea?

ea4k, 2021-06-18:
Migrating to QSettings has been always in my wishlist but I have not done any test. 
I think that keeping a plain snd human readable text file with the config is good for the user.
Would this be possible using QSettings?

foldynl, 2021-06-18:
Yeah, that's also my philosophy. I am not a fan of Win's registry etc. My idea is to use an init format (text format) and use the same directory where current klogrc is stored. Unfortunately, I am not sure whether there is a easy way to implement it via QSettings. I will see after few tests.

foldynl, 2021-06-24:
The initial implementation in PR #303