ea4k/klog #315: V1.6 fails to compile on PiOS buster

garethhowell, 2021-07-12:
Downloaded latest tar, 
extracted to temp folder, 
`qmake PREFIX=/usr/local src.pro
make`
resulted in:
`
g++ -c -pipe --coverage -O2 -std=gnu++11 -Wall -W -D_REENTRANT -fPIC -DAPP_VERSION=1.6 -DAPP_NAME=KLog -DAPP_UNIX_NAME=klog -DQT_USE_QSTRINGBUILDER -DQT_STRICT_ITERATORS -DAPP_LINUX -DQT_NO_DEBUG -DQT_PRINTSUPPORT_LIB -DQT_CHARTS_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_SQL_LIB -DQT_NETWORK_LIB -DQT_SERIALPORT_LIB -DQT_CORE_LIB -I. -isystem /usr/include/arm-linux-gnueabihf/qt5 -isystem /usr/include/arm-linux-gnueabihf/qt5/QtPrintSupport -isystem /usr/include/arm-linux-gnueabihf/qt5/QtCharts -isystem /usr/include/arm-linux-gnueabihf/qt5/QtWidgets -isystem /usr/include/arm-linux-gnueabihf/qt5/QtGui -isystem /usr/include/arm-linux-gnueabihf/qt5/QtSql -isystem /usr/include/arm-linux-gnueabihf/qt5/QtNetwork -isystem /usr/include/arm-linux-gnueabihf/qt5/QtSerialPort -isystem /usr/include/arm-linux-gnueabihf/qt5/QtCore -Ibuild/moc -I/usr/lib/arm-linux-gnueabihf/qt5/mkspecs/linux-g++ -o build/obj/hamlibclass.o hamlibclass.cpp
hamlibclass.cpp: In member function ‘rmode_t HamLibClass::mode2HamlibMode(const QString&)’:
hamlibclass.cpp:178:53: warning: unused parameter ‘_m’ [-Wunused-parameter]
 rmode_t HamLibClass::mode2HamlibMode(const QString &_m)
                                      ~~~~~~~~~~~~~~~^~
hamlibclass.cpp: In member function ‘void HamLibClass::init(bool)’:
hamlibclass.cpp:426:21: warning: comparison of integer expressions of different signedness: ‘rig_model_t’ {aka ‘unsigned int’} and ‘int’ [-Wsign-compare]
     if (myrig_model == -1)
         ~~~~~~~~~~~~^~~~~
hamlibclass.cpp:455:88: error: ‘FILPATHLEN’ was not declared in this scope
         strncpy (my_rig->state.rigport.pathname, serialPort.toLocal8Bit().constData(), FILPATHLEN);
                                                                                        ^~~~~~~~~~
hamlibclass.cpp:473:84: error: ‘FILPATHLEN’ was not declared in this scope
     strncpy (my_rig->state.rigport.pathname, serialPort.toLocal8Bit().constData(), FILPATHLEN);
                                                                                    ^~~~~~~~~~
make: *** [Makefile:2443: build/obj/hamlibclass.o] Error 1

`

ea4k, 2021-07-12:
The problem is related to hamlib 4.2.
If it is compiled with a previous hamlib version, it should work.

KLog will be upgraded to hamlib 4.2 as soon as hamlib 4.2 is in Debían testing.

garethhowell, 2021-07-14:
I can confirm that compiling against the default hamlib contained in the pi's buster repo (3.3.5) works perfectly. Thanks.

closed 2021-07-14 by garethhowell