ea4k/klog #339: Flrig is a network device and a serial device in hamlibclass.cpp

w5pny, 2021-08-19:
It seems in hamlibclass.cpp the only network device is netrigctl, but flrig is NOT a serial device,
it is a network device, so choosing flrig fails with hamlib...

w5pny, 2021-10-05:
flrig works in 1.8.5 ONLY if one specifies the address as 127.0.0.1:12345 in the hamlib settings.  This is because
there is no code in hamlibclass.cpp to concatenate the port number onto the address with a ":" in between.
The port number is there as an int, but must be converted to a string and concatenated onto the address.  The settings menu item for the port number is effectively ignored for hamlib in this version of klog.

ea4k, 2021-10-06:
I can't get it work even with your proposal.
The code in master is already concatenating port to address and it makes "Net rig" rig type to work remotelly. 
It is now possible to run rigctld remotelly and connect KLog to that server.

I can't however connect with the FLRig rig type even if I see the right network address:port 

Any idea?

Thanks

commit 485f74dd (2021-10-06): FLRig remote connection is working (closes #339 )