ea4k/klog #347: Log info on top is wrong on log change

ea4k, 2021-08-23:
the log notification at the top of the main window now showing the wrong log after changing logs

ea4k, 2021-08-24:
Bug Major - No log is ever highlighted as current in Settings -> Logs and even selecting one and OK-ing does not then show it as selected on next entering the dialogue. More importantly when changing the log by selecting a different one and OK-ing, the message in the main window header does not change. The log DOES change in the log list but there is no indication anywhere that this has happened.

barjac, 2021-08-29:
This appears to be partly fixed in 1.8.4 - the main window header is now updated instantly. The appropriate line in the Settings->Logs dialogue is still not highlighted unless unless clicked.

commit c2f269a4 (2021-09-03): Hamlib work and fix #347
closed 2021-09-03 by ea4k