ea4k/klog #352: WSJTX and KLog can't get data from radio at the same time.

ea4k, 2021-08-25:
It seems that WSJTX can only get data from the radio if hamlib is disabled in klog. WSJTX runs fine and klog runs fine, but not together.

 I have never seen this one before. Hamlib is 4.2.0 which is default in Mga8 and was used for the build. I will uninstall and test with the official Mga8 package (1.4.7).

Yes 1.4.7 works perfectly. No Issues with hamlib at all so this is a bug in klog-1.8.2/3

ea4k, 2021-08-25:
Hamlib error: Protocol error
kenwood_open: cannot get identification
kenwood.c(873):kenwood_open return(-8)
rig.c(1000):rig_open return(-8) while opening connection to rig

barjac, 2021-08-29:
Bug is in klog > 1.5.3 and still present in 1.8.4

barjac, 2024-01-12:
This can be closed as old.