ea4k/klog #359: Clublog: Log upload fails with error 201.

ea4k, 2021-08-29:
[image]

ea5wa, 2021-08-29:
After making a new QSO, I tried to upload this new QSO to Clublog with the option in "Tools-Upload QSOs to Clublog" and again error 201 message

ea4k, 2021-09-27:
Can you please check it again?
Please tell me current version and operating system.

It is working in my KLog.

ea5wa, 2021-09-27:
Testing it with your master (1.8.5) works fine. Testing it again with 1.8.4 (windows 11) I have still same error message

ea5wa, 2021-09-27:
Tested in two different windows laptops (one with w10 and the other with w11) with 1.8.4 version, same error message

ea4k, 2021-10-03:
Not able to reproduce the error in KLog 1.8.5

ea4k, 2021-10-03:
I close as I can't reproduce

closed 2021-10-03 by ea4k