ea4k/klog #390: New widget for QRZ.com search

ea4k, 2021-11-27:
Add a widget that allows the user to search for calls in qrz.com without entering the call in the qso input widget

barjac, 2021-12-04:
No this was not the intention of my comment on Telegram.
When a callsign is entered manually or is transmitted into the callsign field from wsjtx the qrz.com auto lookup fills some other empty fields. This is all fine.
The 'Tools->QRZ.com tools -> Look up the current callsign in QRZ.com' option does nothing visible (maybe it repeats what the auto-lookup has already done).
In this situation that menu option would be more useful if it did a full lookup in a browser.

barjac, 2022-09-22:
Or better still a context menu option when moused over the callsign field to do exactly what the option in the context menu in the log widget does.

ea4k, 2022-09-22:
> No this was not the intention of my comment on Telegram.
> When a callsign is entered manually or is transmitted into the callsign field from wsjtx the qrz.com auto lookup fills some other empty [...]

The Tool does the same than auto-lookup... so yes, maybe adding another tool to open the browser makes sense.

barjac, 2022-12-26:
The critical action that is still missing is to easily search qrz.com in a browser for the callsign in the main window entry field.
How the callsign arrived there is not relevant.
Ideally mousing over the field and using a context menu would be best.

ea5wa, 2023-05-28:
Added this functionality with a Button in mainqsoentrywidget