ea4k/klog #418: Band changes move the frequency to the initial frequency of that band

ea5wa, 2022-01-18:
Before changing the band, KLog should save the frequency in use and come to that frequency the next time this band is selected

barjac, 2022-01-18:
In what mode? Seems rather an odd request to me. Surely the operator operates the radio and the logbook logs.the contacts. What am I missing? Does the rig not save the last frequency used on a band, mine does and it's quite old. With hamlib operating why would you need both rig and klog to remember the last frequency and which would win in the fight for control?

ea5wa, 2022-01-19:
Yes, you are right. When using hamlib should be better ask to the rig in which frequency is, when changing band

ea4k, 2022-01-24:
So, the process would be:
- The user selects is KLog a new band.
- KLog asks the radio to switch to that band without sending any freq?
I don't know if hamlib supports that usecase...but I will check