ea4k/klog #423: QSO editing: change of mode sets RST to default values

cfr34k, 2022-01-24:
When an existing QSO is edited and the mode is changed in the process, the RSTtx and RSTrx data is overwritten with default values.

Steps to reproduce:

- Start KLog (I tried version 1.8.6 and commit edad89242e9b814e344f8b415f5b0c2ee3f03b8b)
- Import a log using _File→Import from ADIF_
- Double-click any entry in the QSO list (_Log_) at the bottom of the window. It will be loaded into the editor above.
- Change the mode.
- The _RST(rx)_ and _RST(tx)_ inputs will be set to the default values for the new mode, for example 59/59 if SSB is selected or 0/0 if FT8 selected.

This automatic modification is easily overlooked (the _QSO_ tab might be in the background!) and potentially corrupts the log entry if it is saved without noticing.

To avoid this, I suggest to never reset the RST values while an existing QSO is edited.

closed 2022-01-25 by ea4k

ea4k, 2022-01-25:
FIxed following the advice. Thank you!

cfr34k, 2022-01-25:
Perfect! Thank you very much for the quick fix :-)

commit 46817195 (2022-01-25): Closes #423