ea4k/klog #436: Klog 1.9 and Hamlib 4.4

n6paz, 2022-02-20:
The latest Klog 1.9 does not communicate using Hamlib 4.4.  No radio communications from logger to the radio via hamlib.   I confirmed other applications are successful using hamlib.  Operation system is Unbuntu 20.04.

ea5wa, 2022-02-21:
I'm just testing with hamlib 4.3.1 (last version in my Ubuntu repository and is working fine

n6paz, 2022-02-21:
OK Juan.

Here is the what I am seeing. I have the released version of Hamlib 4.4 installed. Ubuntu 20.04.03 running. I started hamlib from the command line (see screen shots). In a separate window I test communications with the TS-890 via the network and successfully read the frequency and mode. I run Klog 1.9 and it is unable to read the frequency or mode with the same IP address and port (see screen shot).

73's de N6PAZ, Oscar

[Screenshot at 2022-02-21 13-27-49.png]

[Screenshot at 2022-02-21 13-27-58.png]

[Screenshot at 2022-02-21 13-38-47.png]

[Screenshot at 2022-02-21 13-39-20.png]

Sent from [ProtonMail](https://protonmail.ch), encrypted email based in Switzerland.