ea4k/klog #441: debug log

TuxedoHam, 2022-03-13:
klog 2.0-1 under Linux is writing a debug log file in ~/.klog/ even though it set to "DebugLog=False;" in the config file and the app does not have the check box ticked

ikbenkous, 2023-11-25:
I can confirm that this happens here. klog creates an empty file in `~/.klog/klogdebug.log`. I don't know whether this would be classified as a bug or feature. But it's certainly interesting.

TuxedoHam, 2023-11-26:
> I can confirm that this happens here. klog creates an empty file in `~/.klog/klogdebug.log`. I don't know whether this would be classified as a bug or feature. But it's certainly interesting.
> 
> What do you think should happen? @TuxedoHam No file created at all?

@ikbenkous Well their is a couple options I can see that you can go about this

1. Don't create a debug log what so ever unless it is selected in in the GUI and then when you quit the program it writes out with filename eg time-date_klogdebug.log
2. Hide the debug option in the GUI but have it that the debug is run like `klog --debug` and it writes out to log file named like above and write out when the program is quit either via the GUI or `CTRL + C` in Linux terminal example.
3. I guess you can just leave all the debugging done in the IDE as the developer(s) writing the code.

Option 2 might be the better option due all the different versions of OS's that are around theses days.

ikbenkous, 2023-11-26:
***`~/.klog/klogdebug.log` is not what it seems!***
This file is an ADIF. So either we're both mistaken by the confusing terminology of "log"/"logger"/"logging", or klog is doing something REALLY wrong by saving an ADIF to a log file??

Additionally, the UI says this:
[image]
(Hard to read, I know). It says: Select to use the following name for the logfile without being asked for it again.

What kind of "log" is this? An error log or a ham log?
No idea.

Regarding your proposals:
Only debugging in IDE is a _really_ bad idea. End-users don't have debug symbols, source code, IDEs, skills, time or patience to debug as a developer. Error logging is vital to figure out what happens as an end-user.
Also: log buffers should be flushed every time it's written to. Because if the program crashes before it flushes it out, you can't know what happened right before it crashed.

I'm not fond of spamming "time-date_debug.log" files unless:
1. They're inside a directory in `~/.klog/debug/`
2. Stale log files older than 3 months are automatically deleted/rotated.
3. You can disable automatic deletion/rotation and enable 'spamming log files'

This entire issue has got me a little confused. haha

TuxedoHam, 2023-11-27:
@ikbenkous This is what my `Misc` tab looks like and my QSO's are being logged to a `ADI` file

[image]

The `ADI` file is what I have used from first ever starting to use KLog, so that was/is default file to log QSO to.

As for the `klogdebug.log` seems to be created every time klog is run.