ea4k/klog #448: Too difficult to edit frequency fields.

barjac, 2022-04-26:
It is VERY frustrating when attempting to edit either of the frequency fields.

Try to change 70.000MHz to 70.375.

It appears  impossible until you discover that only by backspacing over one character at a time they can be changed.

So the process to do the above is:
Insert cursor after the first decimal place 0.
Backspace over the 0 and type 3.
Insert cursor after the next 0, backspace over it and type 7.
Insert cursor after the next 0, backspace over it and type 5.
Really!
This is a major bug!
We just want to drop the cursor in the field after the decimal point and type, but this is not allowed!

conorpeterson, 2022-08-17:
This is a duplicate of issue number #378 (https://github.com/ea4k/klog/issues/378)

ea4k, 2023-12-11:
duplicated #378

closed 2023-12-11 by ea4k