ea4k/klog #452: Debian 11 - Compiled from source...  Error in function: bool World::recreate(const QString&)

RudiOnTheAir, 2022-05-11:
Hi

I would like to try klog and made a compilation on Debian11. Added the dependencies as shown in the install instructions.

But @ first start i get the following screen... It seems, i am missing the database. Is that possible??

[image]

ea4k, 2022-05-11:
KLog should create the DB on start, this should be a bug.
I will check and fix, thanks.

RudiOnTheAir, 2022-05-11:
Yes, there are files..

wolf@debian11test:~/.klog$ ls -la
total 288
drwxr-xr-x  2 wolf wolf   4096 May 11 17:15 .
drwxr-xr-x 16 wolf wolf   4096 May 11 17:34 ..
-rw-r--r--  1 wolf wolf 263771 May 11 17:09 cty.csv
-rw-r--r--  1 wolf wolf     77 May 11 17:15 klogdebug.log
-rw-r--r--  1 wolf wolf     25 May 11 17:15 klogrc
-rw-r--r--  1 wolf wolf  12288 May 11 17:15 logbook.dat
wolf@debian11test:~/.klog$

RudiOnTheAir, 2022-05-11:
The content of klogrc:

MainWindowSize=1067x682;

And after the first test i have to delete ~/.klog to start over. Without deleting, nothing happens at start..

RudiOnTheAir, 2022-05-11:
[image]

After that, i get some questions, if i like to keep showing errors, and if i say yes, i see more windows with different "error in funktion" messages.

[image]

[image]

ea4k, 2023-06-26:
It looks like your database is malformed.
It should not be the case unless you moved your DB to another folder or the settings are pointing to another place for the DB.

Is the issue still there? Were you able to fix it?

I will tag It as not

ikbenkous, 2023-11-25:
@RudiOnTheAir ping