ea4k/klog #453: SQL break on "

ea4k, 2022-05-11:
It is needed to clean the data to prevent SQLi
[image]

majkrzak, 2022-05-30:
FYI [this](https://doc.qt.io/qt-5/sql-sqlstatements.html#inserting-updating-and-deleting-records) is the way queries should be build
```c++
QSqlQuery query;
query.prepare("INSERT INTO employee (id, name, salary) "
              "VALUES (:id, :name, :salary)");
query.bindValue(":id", 1001);
query.bindValue(":name", "Thad Beaumont");
query.bindValue(":salary", 65000);
query.exec();
```

ea4k, 2023-06-26:
Absolutely right.
I will start rewriting queries that way.
Thanks