ea4k/klog #460: Fed. Rep. of Germany missing from DXCC list

n6paz, 2022-05-31:
The "Fed. Rep. of Germany" appears to be missing in the "DXCC" tab.   See attached image. 

[image]

closed 2022-06-12 by ea4k

n6paz, 2022-10-11:
​Thank you for your support.

73's de Oscar, N6PAZ

Sent from [ProtonMail](https://protonmail.ch), encrypted email based in Switzerland.

ea4k, 2022-10-11:
What KLog versión are you using?

n6paz, 2022-10-11:
Today I am using version 2.2.1. When the problem was reported it was version 2.0.

The problem described has been fixed in 2.2.1

73's de Oscar, N6PAZ

Sent from [ProtonMail](https://protonmail.ch), encrypted email based in Switzerland.

commit e1f89ca8 (2022-06-12): Closes #460