ea4k/klog #466: Qt 5.15 and deprecated API

sashko, 2022-06-18:
Fails to build with Qt 5.15 which is an LTS release due to the deprecated API:

```
../../src/utilities.cpp: In member function ‘bool Utilities::processConfigLine(const QString&)’:
../../src/utilities.cpp:397:53: error: ‘QStringList QString::split(const QString&, QString::SplitBehavior, Qt::CaseSensitivity) const’ is deprecated: Use Qt::SplitBehavior variant instead [-Werror=deprecated-declarations]
  397 |         QStringList values = line.split("=", QT_SKIP);
      |                                                     ^
In file included from ../../../../opt/Qt/5.15.0/gcc_64/include/QtCore/QString:1,
                 from ../../src/utilities.h:37,
                 from ../../src/utilities.cpp:27:
../../../../opt/Qt/5.15.0/gcc_64/include/QtCore/qstring.h:604:17: note: declared here
  604 |     QStringList split(const QString &sep, SplitBehavior behavior,
      |                 ^~~~~
../../src/utilities.cpp: In member function ‘bool Utilities::isValidVUCCGrids(const QString&)’:
../../src/utilities.cpp:1164:34: error: ‘QStringList QString::split(QChar, QString::SplitBehavior, Qt::CaseSensitivity) const’ is deprecated: Use Qt::SplitBehavior variant instead [-Werror=deprecated-declarations]
 1164 |     tmp << _b.split (',', QT_SKIP);
```

If there's no intention to support older Qt versions, I could probably go on and fix it, or perhaps use ifdefs.

ikbenkous, 2023-11-25:
@sashko `-Werror=deprecated-declarations` is enabled. The compile treats the deprecation warning as an error and stops compilation.
You can probably disable that flag and have it compile without problems.

I don't know what's smart to do here.
Possible solutions:
1. Disable `-Werror=deprecated-declarations` when using Qt 5.15
2. Not fix it and hope it goes away as Qt 5.15 is used less and less.
3. Use generic C++ string functionality in this case (or more often)

@ea4k @ea5wa Can either of you comment on this one?

I suggest not fixing because Qt 5.15 support is valid until:
`2023-05-26 (Qt Legacy licenses) / 2025-05-26 (Qt Subscription license holders only)`

ea4k, 2023-11-26:
Can you explain what do you mean with generic string functionality?
Do you mean for split?

ikbenkous, 2023-11-26:
I meant: replacing `QString` with `std::string`. This will fix the issue locally but is bound to create issues interfacing with the Qt API. Having thought this over I think it's a bad idea.

Even in this case, a custom `split()` function would be required since `std::string` doesn't have a `split_to_vector()` function unlike Qt... or does it? 🤔

closed 2025-10-05 by ea4k