ea4k/klog #468: Segmentation fault (core dumped) when exiting Klog on Ubuntu 22.04

n6paz, 2022-06-22:
Segmentation fault (core dumped) when exiting Klog on Ubuntu 22.04

$ klog
QObject::connect(QQuickWindow, QDeclarativeGeoMap): invalid nullptr parameter
Segmentation fault (core dump)

See attached core file.

73's de. Oscar N6PAZ

[_usr_bin_klog.1000.crash.gz](https://github.com/ea4k/klog/files/8960440/_usr_bin_klog.1000.crash.gz)
)

ea4k, 2022-06-23:
Thanks for the report.
Did you compiled you own binary, is it the Ubuntu official one? any other?

n6paz, 2022-06-27:
Q: What version of Klog is crashing?  
A:  V2.1

Q: Did you compile you own binary? 
A: Yes, compiled for Ubuntu 22.04

Q: Is it the Ubuntu official one? anu other?
A: Official release.

Q: What version of QT Packages are you using? 
A: See below:

oalonso@radio:~$ qtdiag
MESA-INTEL: warning: Performance support disabled, consider sysctl dev.i915.perf_stream_paranoid=0

MESA-INTEL: warning: Performance support disabled, consider sysctl dev.i915.perf_stream_paranoid=0

Qt 5.15.3 (x86_64-little_endian-lp64 shared (dynamic) release build; by GCC 11.2.0) on "xcb"
OS: Ubuntu 22.04 LTS [linux version 5.15.0-40-generic]

Architecture: x86_64; features: SSE2 SSE3 SSSE3 SSE4.1 SSE4.2 AVX AVX2

Environment:

Features: QT_NO_EXCEPTIONS

Library info:
  PrefixPath: /usr
  DocumentationPath: /usr/share/qt5/doc
  HeadersPath: /usr/include/x86_64-linux-gnu/qt5
  LibrariesPath: /usr/lib/x86_64-linux-gnu
  LibraryExecutablesPath: /usr/lib/x86_64-linux-gnu/qt5/libexec
  BinariesPath: /usr/lib/qt5/bin
  PluginsPath: /usr/lib/x86_64-linux-gnu/qt5/plugins
  ImportsPath: /usr/lib/x86_64-linux-gnu/qt5/imports
  Qml2ImportsPath: /usr/lib/x86_64-linux-gnu/qt5/qml
  ArchDataPath: /usr/lib/x86_64-linux-gnu/qt5
  DataPath: /usr/share/qt5
  TranslationsPath: /usr/share/qt5/translations
  ExamplesPath: /usr/lib/x86_64-linux-gnu/qt5/examples
  TestsPath: /usr/tests
  SettingsPath: /etc/xdg

Standard paths [*...* denote writable entry]:
  DesktopLocation: "Desktop" */home/oalonso/Desktop*
  DocumentsLocation: "Documents" */home/oalonso/Documents*
  FontsLocation: "Fonts" */home/oalonso/.local/share/fonts* /home/oalonso/.fonts /home/oalonso/.local/share/flatpak/exports/share/fonts /var/lib/flatpak/exports/share/fonts /usr/local/share/fonts /usr/share/fonts /var/lib/snapd/desktop/fonts
  ApplicationsLocation: "Applications" */home/oalonso/.local/share/applications* /home/oalonso/.local/share/flatpak/exports/share/applications /var/lib/flatpak/exports/share/applications /usr/local/share/applications /usr/share/applications /var/lib/snapd/desktop/applications
  MusicLocation: "Music" */home/oalonso/Music*
  MoviesLocation: "Movies" */home/oalonso/Videos*
  PicturesLocation: "Pictures" */home/oalonso/Pictures*
  TempLocation: "Temporary Directory" */tmp*
  HomeLocation: "Home" */home/oalonso*
  AppLocalDataLocation: "Application Data" */home/oalonso/.local/share/QtProject/qtdiag* /home/oalonso/.local/share/flatpak/exports/share/QtProject/qtdiag /var/lib/flatpak/exports/share/QtProject/qtdiag /usr/local/share/QtProject/qtdiag /usr/share/QtProject/qtdiag /var/lib/snapd/desktop/QtProject/qtdiag
  CacheLocation: "Cache" */home/oalonso/.cache/QtProject/qtdiag*
  GenericDataLocation: "Shared Data" */home/oalonso/.local/share* /home/oalonso/.local/share/flatpak/exports/share /var/lib/flatpak/exports/share /usr/local/share /usr/share /var/lib/snapd/desktop
  RuntimeLocation: "Runtime" */run/user/1000*
  ConfigLocation: "Configuration" */home/oalonso/.config* /etc/xdg
  DownloadLocation: "Download" */home/oalonso/Downloads*
  GenericCacheLocation: "Shared Cache" */home/oalonso/.cache*
  GenericConfigLocation: "Shared Configuration" */home/oalonso/.config* /etc/xdg
  AppDataLocation: "Application Data" */home/oalonso/.local/share/QtProject/qtdiag* /home/oalonso/.local/share/flatpak/exports/share/QtProject/qtdiag /var/lib/flatpak/exports/share/QtProject/qtdiag /usr/local/share/QtProject/qtdiag /usr/share/QtProject/qtdiag /var/lib/snapd/desktop/QtProject/qtdiag
  AppConfigLocation: "Application Configuration" */home/oalonso/.config/QtProject/qtdiag* /etc/xdg/QtProject/qtdiag

File selectors (increasing order of precedence):
  en_US unix linux ubuntu

Network:
  Using "OpenSSL 3.0.2 15 Mar 2022", version: 0x30000020

[... 139 lines cut ...]
  Frames in Flight: 2
  Uniform Buffer Alignment: 64
  Supported MSAA sample counts: 1,2,4,8,16
  Features:
oalonso@radio:~$
    v MultisampleRenderBuffer
    - DebugMarkers
    v Timestamps
    v Instancing
    - CustomInstanceStepRate
    v PrimitiveRestart
    v NonDynamicUniformBuffers
    v NonFourAlignedEffectiveIndexBufferOffset
    v NPOTTextureRepeat
    v RedOrAlpha8IsRed
    v ElementIndexUint
    v Compute
    v WideLines
    v VertexShaderPointSize
    v BaseVertex
    v BaseInstance
    v TriangleFanTopology
    v ReadBackNonUniformBuffer
    v ReadBackNonBaseMipLevel
  Texture formats: RGBA8 BGRA8 R8 R16 RED_OR_ALPHA8 RGBA16F RGBA32F R16F R32F D16 D32F BC1 BC2 BC3 BC4 BC5 BC6H BC7 ETC2_RGB8 ETC2_RGB8A1 ETC2_RGBA8 ASTC_4x4 ASTC_5x4 ASTC_5x5 ASTC_6x5 ASTC_6x6 ASTC_8x5 ASTC_8x6 ASTC_8x8 ASTC_10x5 ASTC_10x6 ASTC_10x8 ASTC_10x10 ASTC_12x10 ASTC_12x12

ikbenkous, 2023-09-22:
Can confirm this issue. Compiled from source.
[image]

n6paz, 2023-09-22:
Hello,

I am running KLog 2.3.2 compiled from source code. KLog produces the following error when I exit the problem normally (see below).

73's de Oscar, N6PAZ

[Screenshot at 2023-09-21 18-59-34.png]

[Screenshot at 2023-09-21 18-56-05.png]

Sent from [ProtonMail](https://protonmail.ch), encrypted email based in Switzerland.