ea4k/klog #474: KLog 2.1 crashes when selecting Tools/Show Maps on Ubuntu 22.04

ftomasczik, 2022-07-17:
I have **klog 2.1** compiled on **Ubuntu 22.04** (64 bits) and when I select **Tools** and then **Show Maps** it crashes.

The error on my console is:

```
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
QIODevice::write (QFile, "/home/federico/.klog/klogdebug.log"): device not open
QIODevice::write (QFile, "/home/federico/.klog/klogdebug.log"): device not open
Warning: QQml_colorProvider: no color provider has been set!
qrc:qml/mapqmlfile.qml:136:24: Invalid property assignment: color expected 
                     color: "black" 
                            ^
Violación de segmento (`core' generado)
```
Thanks a lot!
73 de LW3ESH

dfscott2011, 2023-01-16:
I have klog 2.3 compiled on Ubuntu 22.04. When I select Tools > Show Map, klog locks up then gives an option box, "KLog" is not responding, you may choose to wait or close program.
Dropdowns on map box are not clickable
I do have a contact displayed, and the map displays with 'Rolla'.
Happy to provide anything else

Thank you for the program, it works very well.
N0DFS

muihlinn, 2023-02-17:
On 20.04 just does segfault:

qrc:qml/mapqmlfile.qml:47:32: Syntax error
Segmentation fault (core dumped)

geogarden, 2023-08-06:
KLog is not responding, you may choose to wait or close program. Main ubuntu 23.04

ikbenkous, 2023-11-25:
I can confirm this bug as well. Do note that it only seems to hang AFTER it renders the map and starts becoming unresponsive _after zooming/dragging the map!_

alandove, 2024-02-27:
I can also confirm this: KLog 2.3.4 on Ubuntu 22.04, choose Tools > Show Map, and the application locks up. Ubuntu gives the "Wait or Force Quit" pop-up. Choosing "Wait" just closes the pop-up for a few seconds, then it comes back. KLog is completely unresponsive.

closed 2025-05-03 by ea4k