ea4k/klog #502: Upload to QRZ or Clublog

TuxedoHam, 2022-10-02:
Hi All,

I have an issue uploading to QRZ and Clublog without marking all my log with "Queue all the QSOs to be uploaded", the way I have been getting around this in some way is to go through that and only upload the date range I wish to upload that has not been added to QRZ and Clublog.

I never had this with Devuan (Debain fork), I have recently moved this laptop back to Debian (testing) and the issue has started with 2.1.1, I have gone through the fresh install process and re imported my log etc, I had the issue with 2.2.1 from unstable.

ea4k, 2022-10-05:
Can you please reword the issue? I am not able to understand the issue. 
Thank you.

TuxedoHam, 2022-10-11:
I have an issue uploading new QSO's to QRZ and Clublog. I can't queue new QSO's (that haven't previously been uploaded) on their own. The only way I can is to queue all (including ones already uploaded), then filter that by date/time so that only the QSO's that weren't previously uploaded are queued.

This worked as expected in 2.1.1, but since switching from Devuan to Debian I'm using 2.2.1 which is showing this bug.

commit 3da4c4d8 (2023-04-10): Closes #502, #542, #526, #528, #450, #372, #350
closed 2023-04-10 by ea4k