ea4k/klog #504: Build error -lhamlib missing, hamlib installed

clytle374, 2022-10-04:
Sorry if this has a simple answer, but I usually build software with cmake and qmake is new to me.  cmake-gui usually gets me out of these issues, but this is a build time error.  I will gladly provide any requested information.  
` cannot find -lhamlib: No such file or directory`

I'm on a gentoo linux system.  So building from source is 'the way' software is installed. Installing 'packages' can cause big trouble.  I'm including my hamlib version and options.  klog is from git commit 6bd92c3a490bbd8592aa7b9a3c0be75f71b2af24 

qmake info
`Project MESSAGE: Building KLog 2.3~RC1
Project MESSAGE: Qt 5.15.5 in /usr
Project MESSAGE: Sources
Project MESSAGE: Other files
Project MESSAGE: More...
Project MESSAGE: Translations
Project MESSAGE: End of translations...
Project MESSAGE: unix:!mac
`
hamlib info
`[ebuild   R    ] media-libs/hamlib-4.4:0/4.2::gentoo  USE="-doc -perl -python -tcl" PYTHON_SINGLE_TARGET="python3_10 -python3_8 -python3_9"`

Thanks, Cory

Build failure
`g++ -Wl,-O1 -o build/target/klog build/obj/main.o build/obj/aboutdialog.o build/obj/awardswidget.o build/obj/statsfieldperbandwidget.o build/obj/elogqrzlog.o build/obj/eqslutilities.o build/obj/lotwutilities.o build/obj/mainqsoentrywidget.o build/obj/mainwindow.o build/obj/mainwindowinputqso.o build/obj/mainwindowinputcomment.o build/obj/mainwindowmydatatab.o build/obj/mainwindowinputothers.o build/obj/mainwindowinputeqsl.o build/obj/mainwindowinputqsl.o build/obj/mainwindowsattab.o build/obj/qso.o build/obj/searchmodel.o build/obj/searchwindow.o build/obj/setupdialog.o build/obj/hamlibnetworkconfigwidget.o build/obj/hamlibserialconfigwidget.o build/obj/setuppageelog.o build/obj/setuppagelogview.o build/obj/setuppagesubdivisionnew.o build/obj/setuppagesubdivisions.o build/obj/adiflotwexportwidget.o build/obj/mapwidget.o build/obj/mapwindowwidget.o build/obj/onlinemessagewidget.o build/obj/showadifimportwidget.o build/obj/showkloglogwidget.o build/obj/logwindow.o build/obj/filemanager.o build/obj/fileawardmanager.o build/obj/database.o build/obj/dataproxy_sqlite.o build/obj/downloadcty.o build/obj/dxcluster.o build/obj/locator.o build/obj/awards.o build/obj/setuppagemisc.o build/obj/setuppageuserdata.o build/obj/setuppagedxcluster.o build/obj/setuppagecolors.o build/obj/setuppagelogs.o build/obj/setuppageworldeditor.o build/obj/setuppagelogsnew.o build/obj/setuppagebandmode.o build/obj/setuppageudp.o build/obj/setuppagesats.o build/obj/setuppagesatsnew.o build/obj/setuppagehamlib.o build/obj/setupentitydialog.o build/obj/startwizard.o build/obj/awarddxmarathon.o build/obj/elogclublog.o build/obj/softwareupdate.o build/obj/softwareupdatedialog.o build/obj/utilities.o build/obj/dxccstatuswidget.o build/obj/logmodel.o build/obj/searchwidget.o build/obj/infowidget.o build/obj/showerrordialog.o build/obj/udpserver.o build/obj/statisticswidget.o build/obj/barchartstats.o build/obj/updatesatsdata.o build/obj/statsgeneralchartwidget.o build/obj/statsdxccsonsatswidget.o build/obj/statsqsosperyearbarchartwidget.o build/obj/statsentitiesperyearbarchartwidget.o build/obj/statscqzperyearbarchartwidget.o build/obj/statsqsospermodebarchartwidget.o build/obj/statsqsosperdxccbarchartwidget.o build/obj/statsqsospercontinentbarchartwidget.o build/obj/statsqsosperhourbarchartwidget.o build/obj/statsqsospermonthbarchartwidget.o build/obj/statsworkedconfirmedpiechartwidget.o build/obj/statsworkedsentpiechartwidget.o build/obj/statssentconfirmedpiechartwidget.o build/obj/statsqsosperbandbarchartwidget.o build/obj/statsgridsonsatswidget.o build/obj/hamlibclass.o build/obj/tipsdialog.o build/obj/world.o build/obj/qrc_klog.o build/obj/moc_setupdialog.o build/obj/moc_aboutdialog.o build/obj/moc_awards.o build/obj/moc_awardswidget.o build/obj/moc_barchartstats.o build/obj/moc_statsfieldperbandwidget.o build/obj/moc_hamlibnetworkconfigwidget.o build/obj/moc_hamlibserialconfigwidget.o build/obj/moc_setuppagelogview.o build/obj/moc_mapwidget.o build/obj/moc_mapwindowwidget.o build/obj/moc_showkloglogwidget.o build/obj/moc_statsgeneralchartwidget.o build/obj/moc_statsdxccsonsatswidget.o build/obj/moc_statsqsosperyearbarchartwidget.o build/obj/moc_statsentitiesperyearbarchartwidget.o build/obj/moc_statscqzperyearbarchartwidget.o build/obj/moc_statsqsosperbandbarchartwidget.o build/obj/moc_statsqsospermodebarchartwidget.o build/obj/moc_statsqsosperdxccbarchartwidget.o build/obj/moc_statsqsospercontinentbarchartwidget.o build/obj/moc_statsqsosperhourbarchartwidget.o build/obj/moc_statsqsospermonthbarchartwidget.o build/obj/moc_statsworkedconfirmedpiechartwidget.o build/obj/moc_statsworkedsentpiechartwidget.o build/obj/moc_statsgridsonsatswidget.o build/obj/moc_statssentconfirmedpiechartwidget.o build/obj/moc_database.o build/obj/moc_dataproxy_sqlite.o build/obj/moc_downloadcty.o build/obj/moc_dxcluster.o build/obj/moc_dxccstatuswidget.o build/obj/moc_elogqrzlog.o build/obj/moc_eqslutilities.o build/obj/moc_hamlibclass.o build/obj/moc_mainwindowinputqso.o build/obj/moc_mainwindowinputcomment.o build/obj/moc_mainwindowmydatatab.o build/obj/moc_mainwindowinputothers.o build/obj/moc_mainwindowinputeqsl.o build/obj/moc_mainwindowinputqsl.o build/obj/moc_mainwindowsattab.o build/obj/moc_lotwutilities.o build/obj/moc_mainqsoentrywidget.o build/obj/moc_mainwindow.o build/obj/moc_qso.o build/obj/moc_searchmodel.o build/obj/moc_searchwindow.o build/obj/moc_logwindow.o build/obj/moc_filemanager.o build/obj/moc_fileawardmanager.o build/obj/moc_startwizard.o build/obj/moc_elogclublog.o build/obj/moc_softwareupdate.o build/obj/moc_softwareupdatedialog.o build/obj/moc_utilities.o build/obj/moc_logmodel.o build/obj/moc_searchwidget.o build/obj/moc_infowidget.o build/obj/moc_showerrordialog.o build/obj/moc_statisticswidget.o build/obj/moc_setuppagemisc.o build/obj/moc_setuppageuserdata.o build/obj/moc_setuppagedxcluster.o build/obj/moc_setuppagecolors.o build/obj/moc_setuppagelogs.o build/obj/moc_setuppageworldeditor.o build/obj/moc_setuppagesats.o build/obj/moc_setuppagesatsnew.o build/obj/moc_setuppagehamlib.o build/obj/moc_setuppagelogsnew.o build/obj/moc_setuppagebandmode.o build/obj/moc_setupentitydialog.o build/obj/moc_setuppageudp.o build/obj/moc_setuppageelog.o build/obj/moc_setuppagesubdivisionnew.o build/obj/moc_setuppagesubdivisions.o build/obj/moc_tipsdialog.o build/obj/moc_udpserver.o build/obj/moc_updatesatsdata.o build/obj/moc_adiflotwexportwidget.o build/obj/moc_onlinemessagewidget.o build/obj/moc_showadifimportwidget.o build/obj/moc_world.o   -lhamlib /usr/lib64/libQt5PrintSupport.so /usr/lib64/libQt5Charts.so /usr/lib64/libQt5QuickWidgets.so /usr/lib64/libQt5Widgets.so /usr/lib64/libQt5Quick.so /usr/lib64/libQt5Gui.so /usr/lib64/libQt5Sql.so /usr/lib64/libQt5QmlModels.so /usr/lib64/libQt5Qml.so /usr/lib64/libQt5Network.so /usr/lib64/libQt5SerialPort.so /usr/lib64/libQt5Positioning.so /usr/lib64/libQt5Core.so -lGL -pthread   
/usr/lib/gcc/x86_64-pc-linux-gnu/11.3.0/../../../../x86_64-pc-linux-gnu/bin/ld: cannot find -lhamlib: No such file or directory`

ea4k, 2022-10-05:
You need the development package of hamlib in your system and other dependencies before you try to compile KLog.

Please check for a Linux help.

https://github.com/ea4k/klog/raw/master/src/INSTALL-linux

Please join the telegram channel as that will be the easiest way to get support. 
https://t.me/klogchat

clytle374, 2022-10-07:
I did read the manual.  I've never had to instal a development package before on gentoo.   I'm installing telegram and its dependants now. 

Just for reference here, I got the program to build and start.  I had to hack the makefile to add the hamlib paths.   I was not able to figure out how to get qmake to find the files.  So ugly hack like this

`INCPATH       = -I../sources/klog/src -I. -I/usr/include/qt5 -I/usr/include/qt5/QtPrintSupport -I/usr/include/qt5/QtCharts -I/usr/include/qt5/QtQuickWidgets -I/usr/include/qt5/QtWidgets -I/usr/include/qt5/QtQuick -I/usr/include/qt5/QtGui -I/usr/include/qt5/QtSql -I/usr/include/qt5/QtQmlModels -I/usr/include/qt5/QtQml -I/usr/include/qt5/QtNetwork -I/usr/include/qt5/QtSerialPort -I/usr/include/qt5/QtPositioning -I/usr/include/qt5/QtCore -Ibuild/moc -I/usr/include/libdrm -I/usr/lib64/qt5/mkspecs/linux-g++ -I/usr/include/hamlib -I/usr/include/hamlib/ampclass.h -I/usr/include/hamlib/amplifier.h -I/usr/include/hamlib/amplist.h -I/usr/include/hamlib/rig.h -I/usr/include/hamlib/rig_dll.h -I/usr/include/hamlib/rigclass.h -I/usr/include/hamlib/riglist.h -I/usr/include/hamlib/rotator.h -I/usr/include/hamlib/rotclass.h -I/usr/include/hamlib/rotlist.h`

and this
`LIBS          = $(SUBLIBS)  /usr/lib64/hamlib/libhamlib++.so /usr/lib64/hamlib/libhamlib.so /usr/lib64/libQt5PrintSupport.so /usr/lib64/libQt5Charts.so /usr/lib64/libQt5QuickWidgets.so /usr/lib64/libQt5Widgets.so /usr/lib64/libQt5Quick.so /usr/lib64/libQt5Gui.so /usr/lib64/libQt5Sql.so /usr/lib64/libQt5QmlModels.so /usr/lib64/libQt5Qml.so /usr/lib64/libQt5Network.so /usr/lib64/libQt5SerialPort.so /usr/lib64/libQt5Positioning.so /usr/lib64/libQt5Core.so -lGL -pthread`

ea4k, 2022-10-08:
KLog does not need perl or python to build, maybe they are used by hamlib.

Did you run qmake.src?

clytle374, 2022-10-09:
I ran 
`qmake PREFIX=/usr/local src.pro`
then 
`make`

perl and python are build options for the hamlib build.  It's just something with qmake not finding hamlib as if I hack the makefile it will build and run.  I found the list of files in libhamlib-dev for Debian 

`/usr/include/hamlib/amplifier.h
/usr/include/hamlib/amplist.h
/usr/include/hamlib/rig.h
/usr/include/hamlib/rig_dll.h
/usr/include/hamlib/riglist.h
/usr/include/hamlib/rotator.h
/usr/include/hamlib/rotlist.h
/usr/lib/x86_64-linux-gnu/libhamlib.a
/usr/lib/x86_64-linux-gnu/libhamlib.so
/usr/lib/x86_64-linux-gnu/pkgconfig/hamlib.pc
/usr/share/aclocal/hamlib.m4
/usr/share/doc/libhamlib-dev/changelog.Debian.amd64.gz
/usr/share/doc/libhamlib-dev/changelog.Debian.gz
/usr/share/doc/libhamlib-dev/changelog.gz
/usr/share/doc/libhamlib-dev/copyright
`
Then comparing it to the files installed by hamlib on my system.  

` * Contents of media-libs/hamlib-4.4:
/etc
/etc/env.d
/etc/env.d/73hamlib
/usr
/usr/bin
/usr/bin/ampctl
/usr/bin/ampctld
/usr/bin/rigctl
/usr/bin/rigctlcom
/usr/bin/rigctld
/usr/bin/rigmem
/usr/bin/rigsmtr
/usr/bin/rigswr
/usr/bin/rigtestlibusb
/usr/bin/rotctl
/usr/bin/rotctld
/usr/include
/usr/include/hamlib
/usr/include/hamlib/ampclass.h
/usr/include/hamlib/amplifier.h
/usr/include/hamlib/amplist.h
/usr/include/hamlib/rig.h
/usr/include/hamlib/rig_dll.h
/usr/include/hamlib/rigclass.h
/usr/include/hamlib/riglist.h
/usr/include/hamlib/rotator.h
/usr/include/hamlib/rotclass.h
/usr/include/hamlib/rotlist.h
/usr/lib64
/usr/lib64/hamlib
/usr/lib64/hamlib/libhamlib++.so -> libhamlib++.so.4.0.4
/usr/lib64/hamlib/libhamlib++.so.4 -> libhamlib++.so.4.0.4
/usr/lib64/hamlib/libhamlib++.so.4.0.4
/usr/lib64/hamlib/libhamlib.so -> libhamlib.so.4.0.4
/usr/lib64/hamlib/libhamlib.so.4 -> libhamlib.so.4.0.4
/usr/lib64/hamlib/libhamlib.so.4.0.4
/usr/lib64/hamlib/pkgconfig
/usr/lib64/hamlib/pkgconfig/hamlib.pc
/usr/lib64/pkgconfig
/usr/lib64/pkgconfig/hamlib.pc
/usr/share
/usr/share/aclocal
/usr/share/aclocal/hamlib.m4
/usr/share/doc
/usr/share/doc/hamlib-4.4
/usr/share/doc/hamlib-4.4/AUTHORS.bz2
/usr/share/doc/hamlib-4.4/COPYING.LIB.bz2
/usr/share/doc/hamlib-4.4/COPYING.bz2
/usr/share/doc/hamlib-4.4/ChangeLog.bz2
/usr/share/doc/hamlib-4.4/LICENSE.bz2
/usr/share/doc/hamlib-4.4/NEWS.bz2
/usr/share/doc/hamlib-4.4/PLAN.bz2
/usr/share/doc/hamlib-4.4/README.betatester.bz2
/usr/share/doc/hamlib-4.4/README.bz2
/usr/share/doc/hamlib-4.4/README.developer.bz2
/usr/share/man
/usr/share/man/man1
/usr/share/man/man1/ampctl.1.bz2
/usr/share/man/man1/ampctld.1.bz2
/usr/share/man/man1/rigctl.1.bz2
/usr/share/man/man1/rigctlcom.1.bz2
/usr/share/man/man1/rigctld.1.bz2
/usr/share/man/man1/rigmem.1.bz2
/usr/share/man/man1/rigsmtr.1.bz2
/usr/share/man/man1/rigswr.1.bz2
/usr/share/man/man1/rotctl.1.bz2
/usr/share/man/man1/rotctld.1.bz2
/usr/share/man/man7
/usr/share/man/man7/hamlib-primer.7.bz2
/usr/share/man/man7/hamlib-utilities.7.bz2
/usr/share/man/man7/hamlib.7.bz2
`
There isn't a specific package for dev as the system always contains the complete files and build system for all packages, since all packages are built from source.  It's probably as simple as the location of the files.  debian uses /usr/lib and gentoo /usr/lib64.  I haven't found any way to point qmake to the files.   I tried using qtcreator and was just as lost.  Hacking the makefile isn't a good option either as it's overwritten every time.  Maybe this gives you an idea?  

Thanks
Cory

clytle374, 2022-10-13:
Do I need to be looking to qmake to resolve this?  If so you have any suggestions on groups?  

It's obviously just a config error.  But I don't know the qmake system enough to have a clue how to configure it.  

Hamlib has options for perl and python being built in.  But it's not clearly not needed for klog.

Thanks
Cory

clytle374, 2022-10-18:
I build and test Kicad, Freecad, SDR++, Prusaslicer, and a couple other software projects.  
Never had such a seemingly simple configuration error stump me.  Just adding the paths to the makefile fix it.  I get a bunch of warnings that entries like /usr/include/hamlib/rotator.h are not a path but it builds and runs. And runs in place so I don't have to install it into my system.

clytle374, 2022-10-18:
I build and test Kicad, Freecad, SDR++, Prusaslicer, and a couple other software projects.
Never had such a seemingly simple configuration error stump me. Just adding the paths to the makefile fix it. I get a bunch of warnings that entries like /usr/include/hamlib/rotator.h are not a path but it builds and runs. If I remove them all the build fails. It even runs in place so I don't have to install it into my system, which is great.
But I've spent too much time one this.  Closing and abandoning.

closed 2022-10-18 by clytle374