ea4k/klog #509: Mode: DMR

TuxedoHam, 2022-10-16:
Under `Settings => Bands/Modes` any chance having the Mode of `DMR` added. As a added bonus for us that have hotspots would there be a way to the option to be able to have them as a programmable option so that they can be easily access via drop down menu or something without having to add the frequencies for the hotspot all the time due no hamlib support or HT operation.

Example:
Due to that it is JOTA/JOTI weekend I finding it to be a task to having to input DMR duplex hotspot frequencies in, in a timely manner.

ea4k, 2022-10-16:
Add the mode/sumbode added in ADIF 3.1.3
Changes to [Mode](http://www.adif.org/313/ADIF_313.htm#Mode_Enumeration) and [Submode](http://www.adif.org/313/ADIF_313.htm#Submode_Enumeration) enumerations:

Addition of new Submodes C4FM, DMR, & DSTAR to Mode DIGITALVOICE and deprecation of C4FM and DSTAR as Modes.
Addition of new Mode DYNAMIC and Submodes VARA HF, VARA SATELLITE, VARA FM 1200, & VARA FM 9600.

ea4k, 2022-10-16:
Closed in KLog-2.3-RC3

Please open a new issue with all about the hotspots.

closed 2022-10-16 by ea4k