ea4k/klog #524: upload the queued QSOs to Clublog does not work after upgrading to v2.3 on linux

m0wtn, 2022-10-30:
After selecting upload queued QSOs to clublog there are no QSOs listed for upload.  The dropdown lists for Station callsign and My Locator only contain Not defined and quantity of QSOs states zero even though there are several QSOs within my log that have the ClubLog field set to "M - Modified".

m0wtn, 2022-10-31:
More info: Today I noticed that before I try and upload any contacts to eQSL, LoTW or ClubLog then I can use the upload queued QSOs to Clublog and I can see that the Station callsign dropdown list is set to my callsign and the correct quantity of QSOs are listed for upload to Clublog.  Once I have uploaded the queued QSOs to LoTW and then I try and use the upload the queued QSOs to Clublog then the only value within the Station callsign dropdown list is "Not defined" and my QSOs are not listed any more.

ea4k, 2022-11-05:
I am trying to reproduce your error but it is working ok in 2.3.
Can you please follow this procedure:
- Export your log to ADIF.
- Check that the QSOs that you are willing to update have:
   - <STATION_CALLSIGN defined for your call.
   - <CLUBLOG_QSO_UPLOAD_STATUS:1>M

m0wtn, 2022-11-05:
Here is an example QSO that I just exported from klog:
<CALL:6>EA4HTI <QSO_DATE:8>20221105 <TIME_ON:6>152930 <FREQ:7>28.0751 <BAND:3>10M <FREQ_RX:7>28.0751 <BAND_RX:3>10M <MODE:3>FT8 <GRIDSQUARE:4>IM68 <MY_GRIDSQUARE:6>IO82XR <STATION_CALLSIGN:5>M0WTN <QSO_DATE_OFF:8>20221105 <TIME_OFF:6>153030 <CQZ:2>14 <ITUZ:2>37 <DXCC:3>281 <CONTACTED_OP:0> <EQ_CALL:0> <EQSL_QSL_SENT:1>Q <HRDLOG_QSO_UPLOAD_STATUS:1>M <LOTW_QSL_SENT:1>Q <CLUBLOG_QSO_UPLOAD_STATUS:1>M <QRZCOM_QSO_UPLOAD_STATUS:1>M <OPERATOR:5>M0WTN <OWNER_CALLSIGN:0> <QSL_VIA:1>B <RST_SENT:3>+13 <RST_RCVD:3>-13 <TX_PWR:2>25 <APP_KLOG_LOGN:1>3 <EOR>

You can see that the STATION_CALLSIGN is set to my callsign and the CLUBLOG_QSO_UPLOAD_STATUS is set to 'M'.

If I upload this QSO to LoTW then the LOTW_QSL_SENT changes from 'Q' to 'Y'.  If we now use the klog "upload the queued QSOs to Clublog" function then the above QSO is not listed as available to send to Clubblog.  If we now change the LOTW_QSL_SENT field back from 'Y' to 'Q' then we can now see that the above QSO is listed as available to send to Clubblog.  From my testing it appears that any records that have the CLUBLOG_QSO_UPLOAD_STATUS set to 'M' will be listed as available to send to Clublog but only if one or more of these records also has the LOTW_QSL_SENT field set to 'Q'.  This means that if we upload new QSOs to LoTW before we attempt to upload to Clublog and we instruct klog to update all records sent to LoTW then there will not be any QSOs listed as available to send to Clubblog.

m0wtn, 2023-08-06:
This is no longer an issue since I upgraded to klog v2.3.1

closed 2023-08-06 by m0wtn