ea4k/klog #529: Improve TX and RX frequency input

m0wtn, 2022-11-30:
It is currently difficult to manually enter TX and RX frequencies into the QSO tab within the main window.  If I highlight the current frequency and start typing a new frequency only the digits before the decimal point are accepted (MHz).  To enter the digits after the decimal point (kHz) I need to highlight (select) a single digit and then type in a new digit.  I need to do this for every/all digits after the decimal point.  When frequencies are automatically being populated by WSJT-X for example then this is not an issue but if the user is entering all data manually, such as when manually logging voice contacts then this is a problem and wastes a lot of the users time.  Could the frequency fields allow a complete frequency, such as "14.263" to be directly entered with the keyboard?

Also the up/down arrows associated with the frequency fields currently increase and decrease the frequency by 1 MHz, would it be possible to change this so that frequencies can instead be increased/decreased by 1 kHz?

conorpeterson, 2023-03-22:
This is a duplicate of issue #378 except for the suggestions on changing the frequency by kHz with the arrow keys.

barjac, 2023-06-14:
Also a duplicate of #448

ea4k, 2023-12-11:
Duplicated #378

closed 2023-12-11 by ea4k