ea4k/klog #543: Unable to upload the Queued/Recent QSO's to Clublog/eQSL after LOTW Upload

har5ha, 2023-01-26:
Hello!,

Thanks! for the wonderful Klog software. 

I upgraded my Klog software today, from Version Klog-2.2.21 to KLog-2.3 and I am unable to upload the Queued/Recent QSO's to Clublog/eQSL after LOTW uploads. 

Step 1) I noticed that while uploading my recent/queued QSO's  I have to select GridLocator(Dropdown) while uploading to LOTW(New Feature). Which it is doing successfully.

Step 2) Return to KLog and try to upload the recent/queued QSO's [same ones from Step 1] ) to Clublog(I don't use realtime upload option) or) eQSL. The window that opens doesn't pick up my Callsign or my Grid Locator, and the recent/queued QSO's  list remains empty. So I am unable to upload to Clublog or eQSL.

I believe this has something to do with the GridLocator option/code that was added in KLog-2.3.

Thanks!

m0wtn, 2023-02-03:
This is the same issue that I reported on October 30th #524.  To workaround this issue I currently use klog to upload to Clublog before I use klog to upload to LoTW..

ea5wa, 2023-02-18:
I don't really understand the problem. I should upload QSOs daily first LoTW then ClubLog and finally eQSL without any problem

m0wtn, 2023-03-07:
I believe that you do not see the problem because you probably have multiple QSOs that can be uploaded.  The easiest way to reproduce the problem is to modify all of the records in the log table within your sqlite DB so that there are no records available for upload to LoTW (lotw_qsl_sent <> 'Q').  Now add a new QSO into Klog, as default the lotw_qsl_sent field should be set to 'Q' and the clublog_qso_upload_status should be set to 'M'.  If you now upload this QSO to LoTW (first) you will find that you cannot upload the QSO to Clublog because it appears that the "Upload the queued QSOs to ClubLog" function requires there to be at least one record within the log table that has the lotw_qsl_sent field set to 'Q' or else no records are listed as being available to upload to Clublog.

ea5wa, 2023-05-27:
> I believe that you do not see the problem because you probably have multiple QSOs that can be uploaded. The easiest way to reproduce the problem is to modify all of the records in the log table [...]

I did what you said and, I uploaded first to loTW correctly. After that I uploaded the same QSO to ClubLog and eQSL without any problem