ea4k/klog #564: Download of cty.csv failed with the following error code: TLS initialization failed

fgil, 2023-04-12:
Download of cty.csv failed during installation on Windows 10 with "TLS initialization failed"

Also tried Tools > Update cty.csv after installation and got the same error.

Is there a way to manually install the cty.csv file?

OS Information:
Edition	Windows 10 Home
Version	22H2
OS build	19045.2788
Experience	Windows Feature Experience Pack 120.2212.4190.0

[image]

ea4k, 2023-04-12:
Cty.dat can be downloaded from here:

https://www.country-files.com/big-cty/

You should copy the file in the klog folder  and click on update cty.dat

fgil, 2023-04-13:
Pasting the file in the klog folder under the user folder (C:\Users\_username_\klog) worked. Pasting it in the installation folder (C:\Program Files (x86)\KLog) didn't.

Thanks

ea4k, 2023-06-11:
I can't reproduce the message in windows 10, KLog-2.3.1

closed 2023-06-11 by ea4k