ea4k/klog #599: 2.3.1 New Installation Freezes on Splash Screen (Windows & Linux)

cwross, 2023-06-27:
I have installed a fresh copy (no previous databases or installations) on both Windows and Linux, and the program freezes on the KLog splash screen stating "Initializing window..."

On Linux, I compiled from the .tar.gz file provided on Github (Linux Mint 21.1).

Considering this happens on both platforms, it seems like a bug.

Here is the exact sequence of events (same on both platforms):

1. After initial installation, the "Welcome to KLog!" dialog box appears. I click "Next."
2. The "KLog License Information" dialog appears. I click "Acknowledge" and then "Next."
3. The "Country data download" dialog appears. I click "Download." It downloads the file. I click "Finish."
4. A dialog stating "An entities information file (cty.csv) has been detected in your KLog folder and will be loaded" appears - click "OK."
5. The "Settings update" dialog appears, click "OK."
6. A warning stating "It seems there are no QSOs in the database" appears - click "OK."
7. At this point, the application just gets stuck on the "Initializing window..." splash screen with the KLog logo. It gets stuck here indefinitely until the process is killed.

During the initial run, the following is printed to the Linux terminal:

`QSqlQuery::exec: database not open
QSqlQuery::exec: database not open
QObject::connect(QQuickWindow, QDeclarativeGeoMap): invalid nullptr parameter`

9. When run a 2nd time, none of the dialogs appear (as expected), and the program just gets immediately "stuck" on the splash screen. Nothing is printed to the terminal.

cwross, 2023-06-27:
Same issue occurs when built from "bleeding edge" master repo.

cwross, 2023-06-28:
Confirmed that the issue is surrounding the lack of existing log/database. On Linux, I installed the version of KLog supplied by my package manager, 1.8.6. It started up just fine - I created a new log/database, closed the application, uninstalled 1.8.6 using the package manager, then launched 2.3.1: It showed a dialog about updating settings and started normally!

ea4k, 2023-06-28:
Closed in master. Rebuilding for new release. Thank you!

closed 2023-06-28 by ea4k