ea4k/klog #601: 2.3.2 Info Screen Shows Incorrect Contact History

cwross, 2023-06-29:
2.3.2 on Linux Mint

For every entry in my log book (imported via ADIF), the "Info" screen says "Worked but not confirmed" for 6m, 15m, 20m, and 40m even though I have only worked the contact on one band.

In this example, I only worked K7DPS on 40M SSB, never on any other band, but the information is incorrectly displayed.

[image]

Below is the corresponding ADIF record for K7DPS:

`<Call:5>K7DPS
<QSO_Date:8>20230625
<Time_On:6>044920
<Band:3>40M
<Comment:7>ARRL-FD
<N3FJP_COMPUTERNAME:5>RADIO
<Contest_ID:7>ARRL-FD
<Cont:2>NA
<Country:3>USA
<DXCC:3>291
<CQz:2>03
<Class:2>2D
<N3FJP_Initials:3>CWR
<ITUz:2>06
<Mode:3>SSB
<N3FJP_ModeContest:2>PH
<Pfx:2>K7
<QSL_Sent:1>N
<QSL_Rcvd:1>N
<ARRL_Sect:2>AZ
<N3FJP_SPCNum:2>AZ
<State:2>AZ
<N3FJP_StationID:5>RADIO
<N3FJP_TransmitterID:1>0
<eor>`

cwross, 2023-06-30:
Noticed that I can enter ANY US-based callsign -- even one that does not exist in my log file and I have never worked -- and the application tells me that I have "worked but not confirmed" the callsign on  6m, 15m, 20m, and 40m. Example below - KK7XXX does not exist in my log, I just made up the callsign.

[image]

ea5wa, 2023-09-26:
I think "Worked but not confirmed" **is not related to the callsign, is related to that Entity.**

Probably you have QSOs with US stations in those bands (15, 20, 40 and 6 meters)

ea4k, 2024-02-28:
Yes, that message is for the DXCC Entity, not for the specific callsign.

closed 2024-02-28 by ea4k