ea4k/klog #628: Idea: Create 'Frequency' class to store frequency and associated tolerance

ikbenkous, 2023-09-23:
I see that frequencies are handled internally as a `double` representing megahertz.
This may be bad form and open to interpretation. Someone might be under the assumption that those doubles are Hz or kHz instead of MHz.

It _may_ be better to have a dedicated `Frequency` object with a well defined interface.
Things such as:
* overloaded `operator==` to compare two `Frequency` objects that takes into account clock precision.
* `band()` function returning enum. 
* overloaded user literals for nice coding.
* math operations

Something like `f1.band() == f2.band()` is possible, or even math such as `f1 - f2` returning a third frequency.
Division should probably return a double.

Thoughts?

ea4k, 2023-09-23:
I agree, what you says makes sense and would make KLog better designed.
I will add your comment to the Enhancement list.
Thank you

ikbenkous, 2023-09-29:
How would one handle comparing two frequencies taking into account the precision/accuracy/trueness of the transceivers?
For now I think exposing a 'ppm' function and assuming the frequency error is a symmetrical bell curve.

[image]

```C++
(F1 != F2) // true
(F1 == F2) // false

(F1 != F3) // false
(F1 == F3) // true

(F1 != F4) // false // Can we establish that F1 and F4 are not the same? No.
(F1 == F4) // false // Can we establish that F1 and F4 are the same? No!
```

Is this kind of behavior 'too smart'?
Maybe a basic comparison function `isSame(max_diff)` returning true/false is better and less prone to bugs when using...

ea4k, 2024-04-23:
The Frequency class has been created in https://github.com/ea4k/klog/commit/d6b5603c331e86b7fb5277f4613caaa98e720c39

There are some usage already in the DXCluster class.

All contributions are welcome! :-)

closed 2024-04-23 by ea4k