ea4k/klog #630: Define KLOG_DEPRECATED macro?

ikbenkous, 2023-09-29:
How can one mark a function as deprecated?

C++14 and later has a `[[deprecated]]` attribute, but klog is compiled with `-std=gnu++11`.

What about the following in global.h?
```C++
// If C++14
#if __cplusplus >= 201309L
    // use modern deprecation feature
    #define KLOG_DEPRECATED [[deprecated]]
#else
    //... no idea how to handle this, but this seems overly noisy?
    #warning "KLOG_DEPRECATED is not defined yet!"
#endif
```

This way one can mark a function as deprecated as follows:
```C++
KLOG_DEPRECATED void isSameFreq( double x );
```

Thoughts?

ea4k, 2023-10-19:
Why should we mark as deprecated instead of simply removing it?

ikbenkous, 2023-10-20:
Having a way to mark functions as 'deprecated' allows us to keep functions around yet signal to others:  _do not build on this!_ It's a nice feature to have to improve collaboration.

For instance: the `isSameFreq()` should probably be deprecated when the `Frequency` class is finally implemented. Implementing that class is the easy part. Upgrading the code to use that new class might take a while. Hence: the ability to mark `isSameFreq()` as 'deprecated' during that _transition period_ is a smart thing to do.

I'm sure there are (and will be in the future) plenty of cases where deprecating something until someone gets around to remove/rewrite is just a really nice enhancement!

A `KLOG_DEPRECATED` macro **or some other way(!!!) of deprecating things** must be language version agnostic. It should not use language features not available in C++11 until the language version is upgraded.

ea4k, 2024-02-29:
It really makes sense.
Let's work on it.

ea4k, 2024-03-26:
I have added your code to global.h and tested against DataProxy_SQLite::getBands() which is really a deprecated function.
It seems to work.
More tests is recommended.

closed 2024-03-29 by ea4k