ea4k/klog #636: Bandplan per region/dxcc entity

ea4k, 2023-11-05:
Maybe an xml-like file on the internet listing all bandplans so klog can get it and, based on the user dxcc select the right one...

ikbenkous, 2023-11-13:
Maybe bundle some bandplans by default because the internet may not always work (in emergencies or whatever).
You can always update bundles via the internet if newer ones are found.

Some ideas for bundling:
* IARU bandplans (Region 1/2/3)
* Basic allocations per country
* A message that says "Could not check for updated bandplans: internet down." if older than 30 days.

Are there any machine-readable standards for bandplans? Maybe some particular XML formatting?
This stuff oughta be published machine readable somewhere...

ikbenkous, 2023-11-26:
Mockup for a bandplan indicator in the UI:
[image]