ea4k/klog #638: KLog v2.3.2 Crashes on Startup on High Sierra

RobK88, 2023-11-21:
KLog v 2.3.2 crashes on startup on High Sierra.

KLog v2.3 runs fine on High Sierra.

See attached crash log.

[KLog Crash Log.txt](https://github.com/ea4k/klog/files/13429568/KLog.Crash.Log.txt)

ikbenkous, 2023-11-25:
@RobK88 This seems to crash inside of `rig_init()` from hamlib?..
Can you try _**moving**_ your config file in your home directory temporarily to a different place and try starting klog again? This might be a problem with hamlib and your rig.

RobK88, 2023-11-26:
I don't have a rig attached at this time.

But if you look at the attached crash log, it looks like hamlib provided in the KLog application was built for Big Sur.  I will install hamlib built for High Sierra  via MacPorts and replace the hamlib library in the Application with the MacPorts version to see if that fixes the problem.  

KLog v 2.3 still runs on High Sierra.

RobK88, 2023-11-26:
I do not have a rig connected.  Hamlib is NOT activated in the KLog preferences.

I tried temporarily moving `~/.klog/klogrc` to another location and ran KLog.  NO luck.  Application still crashed.

I also tried replacing `libhamlib.4.dylib` inside the application with `libhamlib.4.dylib` built by MacPorts for this Mac running High Sierra.  No success.  Application still crashes at startup.  

If you look at the crash log, you will see:

```
Dyld Error Message:
  Symbol not found: ____chkstk_darwin
  Referenced from: /Applications/KLog.app/Contents/MacOS/../Frameworks/libhamlib.4.dylib (which was built for Mac OS X 11.0)
  Expected in: /usr/lib/libSystem.B.dylib
```

It really looks like the binary is looking for a symbol that is missing on High Sierra.  I suspect KLog was built on a newer Mac where this symbol is present.

KLog v2.3 runs fine on High Sierra.
KLog v2.3.1 and KLog v2.3.2 do not.  They both crash on startup.

I have reverted back to KLog v2.3

RobK88, 2023-11-26:
I also tried to run KLog v2.3.2 on my Mac running Mojave.  It works fine 

I just cannot get KLog v2.3.2 to run on my Mac running High Sierra.

ea4k, 2024-01-01:
I can't build in Sierra but I can help you to build a KLog package from sources in your computer for that version. 
You just need to install Qt and then run a script that will build the KLog d.g package.

closed 2025-05-18 by ea4k