ea4k/klog #657: LOTW upload station callsign "Not defined"

n6paz, 2023-12-29:
Klog version 2.3.3

The LOTW dialog now has Station Callsign "Not defined" and you are unable to make any another selection.   This worked in the previous version and still is function for eQSL and Clublog. 

LOTW:

[image]

Works with eQSL and Clublog: 

[image]

[image]

SpudGunMan, 2024-04-12:
Same issue here

adelman, 2024-05-12:
Can confirm, also happens here.

ea4k, 2025-04-12:
I can confirm the bug.
The issue is because QStringList DataProxy_SQLite::getStationCallSignsFromLogWithLoTWPendingToSend(const int _log)
only return calls if the QSOs are marked as "'Q" (queued) if no QSOs are marked as "Q" no callsign is shown,

commit c83e6014 (2025-04-12): Closes #657
closed 2025-04-12 by ea4k