ea4k/klog #658: module "QtQuick.Controls" is not installed error

n6paz, 2023-12-29:
Compiled from source following all the instruction in the INSTALL-Linux document on Ubuntu 23.10.  Run klog from the shell and get the following messages (see below).  The Klog application seems to be working despite the error messages.   

oalonso@radio:~$ klog
qrc:qml/mapqmlfile.qml:30:1: module "QtQuick.Controls" is not installed 
     import QtQuick.Controls 2.15 
     ^
qrc:qml/mapqmlfile.qml:27:1: module "QtQuick.Window" is not installed 
     import QtQuick.Window 2.12 
     ^
qrc:qml/mapqmlfile.qml:30:1: module "QtQuick.Controls" is not installed 
     import QtQuick.Controls 2.15 
     ^
qrc:qml/mapqmlfile.qml:27:1: module "QtQuick.Window" is not installed 
     import QtQuick.Window 2.12 
     ^

ea4k, 2024-01-24:
Followed the INSTALL.linux in a debian machine and could not reproduce the error.

closed 2024-01-24 by ea4k

ea4k, 2024-01-24:
> Thank you for your support. I followed you suggestion and installed the
> "qml-module-qtquick-window2", then ran Klog (see text below). I still see
> the "module "QtQuick.Controls" is not installed" message. I am running
> Ubuntu 23.10 x86_64 which is based on Debian. Please advise.
>
> 73's de Oscar, N6PAZ
>
> Reading package lists... Done
> Building dependency tree... Done
> Reading state information... Done
> The following NEW packages will be installed:
> qml-module-qtquick-window2
> 0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
> Need to get 25.8 kB of archives.
> After this operation, 134 kB of additional disk space will be used.
> Get:1 http://us.archive.ubuntu.com/ubuntu mantic/universe amd64
> qml-module-qtquick-window2 amd64 5.15.10+dfsg-2 [25.8 kB]
> Fetched 25.8 kB in 0s (104 kB/s)
> Selecting previously unselected package qml-module-qtquick-window2:amd64.
> (Reading database ... 529642 files and directories currently installed.)
> Preparing to unpack
> .../qml-module-qtquick-window2_5.15.10+dfsg-2_amd64.deb ...
> Unpacking qml-module-qtquick-window2:amd64 (5.15.10+dfsg-2) ...
> Setting up qml-module-qtquick-window2:amd64 (5.15.10+dfsg-2) ...
> exit
> failed to create drawable
> qrc:qml/mapqmlfile.qml:30:1: module "QtQuick.Controls" is not installed
> import QtQuick.Controls 2.15
> ^
>
> Sent from [ProtonMail](https://protonmail.ch), encrypted email based in
> Switzerland.

Please run the second apt command that is listed in the INSTALL-linux file.
It should install all dependencies that are needed in execution time, like
the qml modules.

Jaime

n6paz, 2024-01-25:
Jaime,

I followed your instructions and it fixed the "QtQuick.Controls" messages. Thank you. Now is see the following messages:

***@***.***:~/HamRadio/klog-2.3.2/src$ klog
failed to create drawable
QObject::connect(QQuickWindow, QDeclarativeGeoMap): invalid nullptr parameter
QGeoTileProviderOsm: Tileserver disabled at QUrl("http://maps-redirect.qt.io/osm/5.8/satellite")QGeoTileFetcherOsm: all providers resolved

The software appears to work on version 2.3.2, not sure what the errors, warning message mean.

73's de Oscar, N6PAZ

Sent from [ProtonMail](https://protonmail.ch), encrypted email based in Switzerland.