ea4k/klog #677: FO/x calls are not properly identified

ea4k, 2024-01-28:
[image]

ea5wa, 2024-02-02:
getCQzFromPrefix function in dataproxy class is returning "-1" with TX5S (Clipperton Island) although is correctly defined in cty.dat and in prefixesofentity table

ea5wa, 2024-02-02:
[image]

ea5wa, 2024-02-02:
ITU Zone is correctly saved but CQ Zone is failing

ea4k, 2024-02-03:
Seems to identify the DXCC but fails to store that values properly 
[image]

ea5wa, 2024-02-03:
Solved using function world->getEntityCqz(dxcc) in mainwindow

ea5wa, 2024-02-03:
[image]

commit 5952ce57 (2024-02-12): Closes #677 CQ Zone 0 when FO/x
commit b9998eb6 (2024-02-12): Merge pull request #679 from ea5wa/master

Closes #677 CQ Zone 0 when FO/x
closed 2024-02-12 by ea4k