ea4k/klog #700: ADIF import bugs

ea4k, 2024-03-24:
[image]

ea5wa, 2024-03-25:
Maybe the adif is not including that field (EQSL_QSL_SENT) and by default "Q" is used

KB2YSI, 2024-08-21:
I was not able to import ADIF comments on 2.3.3 or 2.3.4.

Sample ADIF (this is a only a single QSO, the real file has 275 QSOs in it):

```
ADIF Export from ADIFMaster v[3.5]
http://www.dxshell.com
Copyright (C) 2005 - 2023 UU0JC, DXShell.com
File generated on 18 Aug, 2024 at 22:16
<ADIF_VER:5>3.1.4
<PROGRAMID:10>ADIFMaster
<PROGRAMVERSION:3>3.5
<EOH>

<CALL:5>AA4TX <MODE:3>SSB <BAND:3>40m <FREQ:5>7.285 <QSO_DATE:8>20240816 <TIME_ON:6>024028 <RST_RCVD:2>55 <RST_SENT:2>53 <STATION_CALLSIGN:6>KB2YSI <MY_GRIDSQUARE:6>FN12nr <DXCC:3>291 <STATE:2>FL <CQZ:1>5 <ITUZ:1>8 <COMMENT:27>POTA US-2133  WWFF KFF-2133 <MY_SIG:4>POTA <MY_SIG_INFO:7>US-2133 <MY_POTA_REF:7>US-2133 <EOR>
```

ea4k, 2025-11-09:
Splited in: #850, #849,#848 .
I will track them and close this one.