ea4k/klog #715: Option to disable the the war-related message

AsciiWolf, 2024-06-18:
Is there any option to disable the war-related message on startup? Don't get me wrong, that war is a horrible thing, but displaying a message about it **every time** KLog gets launched is very, very annoying.

ea4k, 2024-08-27:
The message will be removed in KLog-2.4

closed 2024-08-27 by ea4k

AsciiWolf, 2024-08-27:
Good. But it is still unfortunate that it will be in Ubuntu 24.04 for the next 5 years (or until someone backports the removal).

barjac, 2024-09-30:
> Good. But it is still unfortunate that it will be in Ubuntu 24.04 for the next 5 years (or until someone backports the removal).

In Mageia we patch klog to not show the message as it is against our policy to show such things, maybe you should ask the Ubuntu packager to do the same.

AsciiWolf, 2024-09-30:
Ubuntu packages usually have no maintainers, they are just synced from Debian. It needs to be fixed by some community member and approved by some trusted Ubuntu (Canonical) developer.

barjac, 2024-09-30:
> Ubuntu packages usually have no maintainers, they are just synced from Debian (and this can happen only before new Ubuntu distro release, no bigger updates are usually approved afterwards).

I was not aware that Ubuntu was virtually a clone of Debian! - I have never used it.

> It needs to be fixed by some community member as a "Stable Release Update" and approved by some trusted Ubuntu (Canonical) developer.

Yes once in a stable release it's there for the life of that release in Mageia also, which is why as maintainer I removed the message before our last release.

I wonder when klog-2.4 will be released. Hopefully before our next release. :)

AsciiWolf, 2025-01-02:
@ea4k Any chance the new 3.4 version of KLog can actually be released in the near future so it still has a chance to get into next Debian Stable (that is supported for 5 years) and other important distributions that have new major releases approaching soon? Thanks!

ea4k, 2025-01-02:
I am working in the next version of KLog.
It will come with Qt6 and other important updates.

AFAIK there is no release date for Debian ... so the quick answer would be yes but unfortunately I can't commit as I am coding just in free time :-(

All help closing issues in the milestone will be appreciated. 

https://github.com/ea4k/klog/milestone/16

I will, however, go through the list to think again on the priority of the issues.

Anybody willing to help please focus on bugs.