ea4k/klog #723: CompleteWithPreviousQSO  does not deliver correct data

ea4k, 2024-06-24:


ea5wa, 2024-06-24:
There is an error in QSO::fromDB that gives an error message in QtCreator: **QSqlQuery::value: not positioned on a valid record**

bool QSO::fromDB(int _qsoId)
{
    logEvent (Q_FUNC_INFO, "Start", Debug);
    QString queryString = "SELECT * FROM log WHERE id=:idQSO";
    QSqlQuery query;
    query.prepare(queryString);
    query.bindValue(":idQSO", _qsoId);

variable _qsoId has the correct value (is pointing correctly to the previous QSO with that callsign) but function fromDB is returning bad data

ea4k, 2024-06-24:
The issue is possibly here: qso::fromDB(int _qsoId)

ea5wa, 2024-06-24:
I think the problem is "idQSO" is not a field of that table. The correct field name is "id"

ea5wa, 2024-06-24:
Finally it was a problem with the data inside my logbook.dat. It was filled with erroneous data. 

There is no issue with KLog Code, previousQSO is correctly presenting the data. **You can close this issue**.

Sorry

closed 2024-06-24 by ea4k