ea4k/klog #735: LOTW upload using VARA-HF mode fails.

n6paz, 2024-08-06:
After logging a contact with the mode as "VARA HF" the LOTW fails when uploading complaining about MODE being "DYNAMIC" with the SUBMODE being "VARA HF".   See image below

I exported the log entry and modified the DYNAMIC to DATA and the LOTW system accepted it. 

[image]

ea4k, 2024-08-10:
The option to transform DYNAMIC into DATA before creating the ADIF for LoTW seems to be appropriate.
I will work on that direction but also I will ask in the ADIF.org group to see what other developers are doing just in case there is another trend for this.

https://lotw.arrl.org/lotw-help/frequently-asked-questions/?lang=en#modes

ea4k, 2025-04-14:
A bug has been fixed but the recommendation for this issue in particular is to map the DYNAMIC ADIF mode to DATA in TQSL.

KLog works with ADIF
https://lotw.arrl.org/lotw-help/pref-adi/

commit dd3cd44e (2025-04-14): Closes #735
closed 2025-04-14 by ea4k