ea4k/klog #808: Klog 2.4.1 does not upload  queued QSO onto the LOTW

n6paz, 2025-05-18:
Installed the Klog 2.4.1 on Linux and MacOS.   Everything works as designed, however when I entered a new QSO and selected Tools->LOTW Tools->Upload queued QSOs to LOTW and results in "There are no queued QSO" pop-up dialog box (see below)

[image]

eekdood, 2025-05-30:
Yep. I can confirm your findings are correct. Even though LOTW is marked as queued, I also receive the popup. On my macbook, you can also briefly see the dialog listing the queued QSOs as soon as you clear the error dialog.

Strangely, this didn't happen on QSOs that were directly added from WSJT-x; only those that are imported or manually entered.

ea4k, 2025-05-30:
Please check Settings/Misc/Mark sent eQSL && LoTW in new QSO as queued

I am however working to allow the user to check a default value for one of each of the 4 electronic logs that are currently shown/managed in KLog

eekdood, 2025-06-07:
I can confirm that the setting is enabled to Mark QSOs as queued. Cycling the setting and restarting the application had no effect.

closed 2026-03-19 by ea4k